KoenVanduffel/CR-6_Klipper #3: How configure for dgus_reloaded?

Thinkersbluff, 2022-01-03:
The Readme.md for your fork of DGUS_reloaded points to a specially-modified version of Klipper, but I do not see any guidance on how to merge that modified version with your CR6-specific edits.  

Would you please clarify how to adapt that "special" version of Klipper to work with CR6.cfg and where/how to declare the [display] section?  

Might it help your CR6 project if you also fork that special Klipper variant here, to extend that variant with whatever would correctly interpret a .cfg entry like: 

[display T5Lxx]
printer model: CR6-SE

KoenVanduffel, 2022-02-19:
I have tried to get the CR6 screen to function in Klipper. It is however not the same screen as on the CR10 and I don't have the time and knowledge to do this. Would love if anyone up to the task could do it though.

closed 2022-02-19 by KoenVanduffel

Rukongai, 2022-12-31:
I just got this working. I had to recompile the UI and edit some things in the configs for dgus - but it's working. It only works natively in landscape, so i'll either remount my screen or re-do the UI for vertical orientation

I'll share deets tomorrow when my brain is less fried

Thinkersbluff, 2022-12-31:
Looking forward to reading what you modified to get this working.

I have refactored those "reloaded" screens to run in portrait mode, but I am still chasing-down a couple of run-time bugs that may be specific to my installation or configuration.