KoenVanduffel/CR-6_Klipper #7: Was this ever tested?

oliverhbailey, 2022-09-12:
This is the most disorganized and poorly documented project I have ever seen.  And with a little organization and some decent documentation in addition to people who want to finish what they start, this would be much simpler to load and get running.  I built the firmware and it works.  I actually used the printer.cfg file for the BTT-SKR-CR6 on the site where the printer.cfg files are stored and it worked with a few minor problems.  Then I loaded this garbage and it doesn't work at all.
Instead of walking people through the setup, explain to them what and why they are doing it.  There is so much documentation in so many places and its impossible to tell what os what and which is right.  Ironically, I was looking for two items, the strain gauge procedures,  changing the Z lower stop, and turning the Blue LED on and off.  The lower Z stop is several millimeters too high and it doesn't touch the bed, nor can the stop be changed.  
If something doesn't work, remove it!  If it hasn't been tested in a year, note it!  Otherwise finish it and maintain it.  This no longer works well enough with the CR-6 SE to connect.

KoenVanduffel, 2022-09-12:
WOW what a bucket of acid!!! thanks for that.

First of all this is the configuration I am running myself for over a year now, so yes this has been tested in hundreds of prints. It is in use by hundreds of people on the CR6 community Github. The vast majority of people have installed it without much issues. So seemingly you have a problem reading:
* Which OS: I very clearly state that I go from the FluidPi or MainsailOS packages. If you don't take the effort to get a look there you might have missed that this is simply Raspberry OS.

I never had the intention to explain Klipper, there are tons of websites and YouTubes out there that do it better than I can do. It is merely a walkthrough of 1 particular way to setup Klipper. Looking at your claimed credentials you should have an easy task understanding what happens in the Klipper firmware...

One point you could be right: the optical Z endstop is not used because Klipper does not offer a way to do this (at least without additional coding which is well beyond my knowledge, with your credentials that should be a half hour task I guess). From your comments that is all you have been looking for. Tough luck I did not explicitly write it down.

But now my question: Could you explain me what is disorganized as I far as I see it it is a walktrough for an absolute beginner how to install Klipper. It has a few steps: get FluiddPi or MainsailOS onto a microSD card, tells you the main thing to change in the Raspbian OS, copy the 2 config files Klipper needs, build and flash the firmware itself and set the USB port in the firmware. From there you have a working system. If you just pick a few things and mix it with something else it is up to you to figure out the detail of that.

KoenVanduffel, 2022-09-12:
I just found out that you posted near identical opening sentences on other GitHub pages (probably elsewhere too): "This project is the most disorganized github project I have ever seen." Great effort to find to find the worst one and 3 days later an even worse one :).