KoenVanduffel/CR-6_Klipper #8: CR-6 Max - trying this out

figuerres, 2022-09-19:
just a quick note...

i recently got a new CR-6 Max wit the era board.

i have a pi running with a touch screen and the basic klipper / moonraker / mainsail / klipperscreen loaded and seems to work.

this week i hope to get enough sorted out to where i am brave enough to flash the firmware and get this working.

i will update with anything i learn that may help.

thanks for the start on this!

lamp-town-guy, 2023-03-03:
How did you get it going? My CR6 Max with  CR-ERA V1.1.0.3 doesn't seams to connect to klipper on the SBC. I've tried to compile firmware with way too many config combinations and was unable to connect to it via mainsail.

Tmallard, 2023-03-06:
@figuerres @lamp-town-guy I have two printers with the CR-ERA V1.1.0.3 that have previously run the community FW and now Klipper. Both times I flashed the firmware the Creality board was very picky about which SD card I used. I ended up using a blue EMTEC 8 GB SD card (try one of similar size I cant find a current link). I tried a few smaller ones (128mb, 4 GB) and the boot loader skip right over it even with the Partition set correctly. If the touch screen doesn't freeze then it did not flash Klipper correctly. Also I have always had issues with Mainsail. Local go to the IP address of the PI and you should be good to go. Have fun

lamp-town-guy, 2023-03-06:
You just wrote as I was able to successfully flash it. Right now I think I have everything set up correctly and I'm printing with it already. It required to modify example config because that one was much smaller. I have no problem with sharing it but I'm not using full 400mm^3 volume because my cable chains are not long enough and actually prevent me from doing that. Which is not a problem for me but I wouldn't want to share half baked configs to people.