KoenVanduffel/CR-6_Klipper #9: Recompiled firmware.bin's

LordingtheVoid, 2023-04-22:
Hello! Thanks for your hard work. I've been reading through the guide and about to set up klipper on a 4.5.2 cr 6 SE. It says for convenience in the firmware file I can find these bins. Am I missing something? Because I can only find the configs. I'd hope to save some time setting this up correctly but if I have to compile them myself, I guess I will do that. Thanks!

KoenVanduffel, 2023-04-22:
Hi, Sorry I did not change the text. I removed the precompiled bin files as too often the Klipper codebase changed requiring a new compile which I might or might not have followed quickly.

LordingtheVoid, 2023-04-22:
No problem. Your guide helped a lot! Got it all set up. I probably won't be able to print until Tuesday so I can upgrade a few other parts.

closed 2023-04-22 by LordingtheVoid