KoenVanduffel/CR-6_Klipper #14: Unsafe/Dangerous setting is the config files

spamwax, 2023-12-19:
Please remove this line from [CR6.cfg](https://github.com/KoenVanduffel/CR-6_Klipper/blob/b9cc039d31985bff2748558ae0aa3eb8cdc23653/klipper_config/CR6.cfg#L50C22-L50C22):
`SET_GCODE_OFFSET Z=0`
It essentially overrides user's carefully adjusted setting and very likely causes mechanical damage:

KoenVanduffel, 2023-12-19:
Just to explain why I have it there: The SET_GCODE_OFFSET resets the offset you make babystepping during a print. When the print finished Klipper asks whether you want to save it (add it to the existing Z-offset). If you don't save it it still remembers it for the next print, however if you shutdown or reset the rpi it forgets it. Which is inconsistent.

SET_GCODE_OFFSET=0  does not change your Z-offset. The Z-offset determined before is still applied as always.

Hence if you setup your Klipper instance according the instructions this is quite safe and no different from a simple reboot of the rpi. So the only case where this might cause an issue is upon first setup when not saving the obtained Z-offset value.

Nevertheless i will comment out this line and explain why it can be very handy to have it in here in the comments.

spamwax, 2023-12-25:
Thanks for the explanation. I think I had couple of instances where nozzle was scratching the print bed, probably because after Z-offset calibration, I didn't save/restart.

Just a side question, how did you figure out the pin names (`PB5`, `PA1`, ...) for the Klipper config file? I want to integrate a BLTouch and was hoping to use the current end-stop sensor pin (pictured) and some of the pins used for the display.
[image]

KoenVanduffel, 2023-12-25:
The pin numbers i got from the BTT GitHub of the mainboard https://github.com/bigtreetech/BIGTREETECH-SKR-CR6/blob/master/Hardware/BTT%20SKR-CR6-Pin.pdf
If you run one of the Creality boards they have their own pages, i dont remember where i found those. Probably on Sebastiaan's pages www.damsteen.nl