KoenVanduffel/CR-6_Klipper #16: Issue at the start of 3d  print on CR6-SE

sairfan1, 2024-07-29:
I'm using Creality CR6-SE printer, gcode genreated with Cura 5.6.0

After installation when I upload the file for first print heat bed gets its defined temperature, then nozzle temperature also reach to set temperature, printer starts printing outline of the print, and suddenly nozzle temperature start dropping and I'm getting following error

[image]

as error direct me I set min_extrude_temp in my printer.cfg file restart klipper and send a new print, I did but again i get the same error.

[image]

When print is started, bed temp set to 60C and start raising and then nozzle temp set to 200 and start raising once temp reach to set level print starts then status of nozzle temp shows "off" and Target shows "0" and temp start decreasing, after a while when temp is around 180C I get above error message and status for both bed/nozzle is set to 0 and print stops

[image]

If we look at the klipper log and strange thing is going on, at the point when temperature start dropping target temperature is set to zero "0" I have no idea after setting temperature to 200 why reset to zero, so looked at the gcode file as below

[image]

here are few lines of gcode file generated with cura slicer, here we can see that M104 command was sent to set the temperature followed by start printing commands, I check whole gcode file there no other command issued regarding temperature except that at the end of print to cool down the printer.  

Its been more than a month I'm trying to fix this issue, many restarts, rebuild, and other tries like putting additional command in gcode file to set temperature again etc, but no luck.

KoenVanduffel, 2024-07-30:
It looks like you are not using the start_print macro but rather "manual" start print in the slicer. That is probably where your unknown commands come from as well. These are commands the Klipper does not use.

Somewhere you have an M104, M109 or something else that resets the hotend temperature in the start code. You can make your life a lot easier by using the start macro from the repo and removing anything else from the start code. Same for the end code btw. The start macro takes a few arguments from the slicer to set the correct temperatures and runs the droopline for you.

sairfan1, 2024-08-01:
@KoenVanduffel thanks for getting back on this, you were right that was the issue, I do not have deep knowledge and I was not able to understand that instruction.

closed 2024-08-01 by sairfan1