KoenVanduffel/CR-6_Klipper #17: Extruder Temp to 0 during Probe Tare

gineer, 2024-08-16:
Line 27 of CR6.cfg sets the extruder temperature to 0 as part of [gcode_macro TARE_PROBE].

In line 126 of printer.Creality-4.5.3.cfg you call activate_gcode: TARE_PROBE.

The effect of this is that when the print starts, the bed and extruder heat up, and then the probe is tared, which then also sets the temperature to 0. Shortly after, the printed reports that the extruder temperature is too low and the print stops.

Commenting out line 27 of CR6.cfg resolved this issue for me.

Hope that helps

KoenVanduffel, 2024-08-16:
That is correct, during tarring and probing the heater is switched off. A notorious problem of the CR6 is that the PWM switching of the heater influences the strain gauge leading to inconsistent bed measurements. For that reason the heater is temporarily switched off. If you use the accompanied start print macro the heater is switched back on immediately after probing.

gineer, 2024-08-16:
Thanks for that. I should have RTFIC

closed 2024-08-16 by gineer